grisp/grisp-software #1: Building rtems-libbsd doesn't work from a fresh checkout

peerst, 2016-09-03:
The reason is that there is the sequence

    waf clean
    waf configure

in the build script, but `waf clean` doesn't work if it was never configured.

Workaround:  on the first run after a fresh checkout comment out `waf clean` in 
`build/build-libbsd.sh`

peerst, 2017-03-02:
Didn't run into this one last time I tried.  Is this fixed?

c-mauderer, 2017-03-02:
Only insofar that the "waf clean" is commented. That works fine on a fresh checkout but might make problems on an RTEMS update. It is possible that the libbsd build system does not recognize a changed rtems header correctly and therefore forgets to rebuild some files. So it's now the other way round: It certainly works well on a fresh checkout but it might fail on an update.

peerst, 2017-03-06:
Fixed by checking if wscript exists before calling waf clean

peerst, 2017-05-04:
This fix seems not to have worked, just happened again

c-mauderer, 2017-05-05:
Should be fixed by f40d6fcd65.

closed 2017-05-05 by c-mauderer