grisp/grisp-software #3: Source Builder expects xz at fixed location

c-mauderer, 2016-09-06:
For some reason the rtems-source-builder generates a script with a fixed path to xz. This is a problem on MacOS where xz is located somewhere else.

Excerpt from log:

```
+ build_top=/Users/peer/projects/grisp/grisp-software/rtems-source-builder/rtems/build/arm-rtems4.12-gdb-7.11-x86_64-apple-darwin14.5.0-1
+ gdb_source=gdb-7.11
+ source_dir_gdb=gdb-7.11
+ cd /Users/peer/projects/grisp/grisp-software/rtems-source-builder/rtems/build/arm-rtems4.12-gdb-7.11-x86_64-apple-darwin14.5.0-1
+ /bin/rm -rf gdb-7.11
+ /usr/local/bin/xz -dc /Users/peer/projects/grisp/grisp-software/rtems-source-builder/rtems/sources/gdb-7.11.tar.xz
+ /usr/bin/tar -xvvf -
/Users/peer/projects/grisp/grisp-software/rtems-source-builder/rtems/build/arm-rtems4.12-gdb-7.11-x86_64-apple-darwin14.5.0-1/doit: line 86: /usr/local/bin/xz: No such file or directory
+ cd gdb-7.11
/Users/peer/projects/grisp/grisp-software/rtems-source-builder/rtems/build/arm-rtems4.12-gdb-7.11-x86_64-apple-darwin14.5.0-1/doit: line 87: cd: gdb-7.11: No such file or directory
shell cmd failed: /bin/sh -ex  /Users/peer/projects/grisp/grisp-software/rtems-source-builder/rtems/build/arm-rtems4.12-gdb-7.11-x86_64-apple-darwin14.5.0-1/doit
error: building arm-rtems4.12-gdb-7.11-x86_64-apple-darwin14.5.0-1
```
A call to `/usr/local/bin/xz` is used. But on the affected system xz is located at `/opt/local/bin/xz`:
```
$ type xz
xz is hashed (/opt/local/bin/xz)
```

c-mauderer, 2017-10-25:
Chris Johns added a patch into the RSB that should solve the issue: https://git.rtems.org/rtems-source-builder/commit/source-builder?id=34d7e0cc71b3c416db11d471fe6843e8b7c725ad

Next time the RSB is updated to the latest version, we will get this patch too.

nextl00p, 2018-05-03:
I am affected. I installed xz with mac ports. RTEMS 5 should probably fix this.

c-mauderer, 2018-05-04:
Which branch is affected? The one with update to upstream that Sebastian sent recently or the main branch? The upstream version shouldn't have that issue any more.

nextl00p, 2018-05-04:
Current master. Not yet tested with #42 .

c-mauderer, 2018-05-04:
OK. On master that is still a known bug. I think it's also in the README as a prerequisite.

closed 2018-06-05 by nextl00p