grisp/grisp-software #4: Install missing prerequisite: waf

c-mauderer, 2016-09-06:
The build-scripts should install waf too.

eproxus, 2017-11-27:
Is there a bin folder somewhere in the repo where `waf` could be included? It's just a 100K binary so it could be checked in directly even

c-mauderer, 2017-11-27:
It should be a python script and not even a binary. Maybe we could put it into the `build/src` and copy it from there to the `rtems-install`?

eproxus, 2017-11-27:
True, I meant "executable". I'm not familiar enough with the structure of the rtems build system and this repository to have any opinion about where to place it, other than that checking it in if possible is a nice solution because then it doesn't have to be installed by users

closed 2017-11-28 by c-mauderer

eproxus, 2017-11-28:
Nice! Thanks!

commit 85e7a30e (2017-11-28): build: Add waf to toolchain.

This fixes #4.