grisp/grisp-software #7: RTEMS source builder prerequisites instructions not working

eproxus, 2017-03-01:
The prerequisite *"to check for some of RTEMS source builders prerequisites"* cannot be performed because the folder `rtems-source-builder` is empty upon initial checkout.

eproxus, 2017-03-01:
Just noticed that the check is automatically performed in `./build/build-toolchain.sh`. Perhaps it can be removed from the README?

c-mauderer, 2017-03-02:
Hello @eproxus,

this relates to #5: The check runs a little late in the build process (after the long downloads). It would be better to move it further to the top. I planned to remove the lines from the README as soon as #5 is fixed.

c-mauderer, 2017-11-16:
I think that #5 will need some more time. So I improved the description in the README that it at least works that way.

commit c04dbb09 (2017-11-16): README: Fix description of rsb check.

This fixes #7.
closed 2017-11-16 by c-mauderer