grisp/grisp-software #8: Bootloader doesn't display a version

peerst, 2017-06-01:
The bootloader should have a version number (semver) and display it
on the bootscreen.

Otherwise we can never tell which version of the Bootloader we have on a board

c-mauderer, 2017-06-02:
What kind of version number would you prefer? We could for example use the git-hash or the date of the commit. An alternative would be a manually counted number but I don't like manual numbers because it's easy to forget to change them.

peerst, 2017-06-02:
As mentioned: we use Semantic Versioning in all our projects http://semver.org this is the state of the art of versioning which should be adopted by all sensible projects.

The bootloader should display its version string according to this.  Releases shall be tagged with their version.  Ideally the git hash is shown for intermediate commits which are not released.
Usually I do this with scripts in the build system generating a header file.  This can also help to not forget to bump the version.

Currently we start with a 0.x.y version.

The first bootloader that will be delivered on boards shall have version 1.0.0

peerst, 2017-06-02:
The toplevel repo should in addition be tagged with a tag that includes the bootloader name:

e.g. bootloader-0.1.0

The bootloader repo in addition should be 0.1.0

This is coding standard for all repos that belong to the GRiSP project

c-mauderer, 2017-06-02:
Just a note: The bootloader is not in it's own repo but it is included in the top level repo. So I would suggest to only use the `bootloader-0.1.0` format for the tag.

If it is OK for you I'll tag the bootloader for the first V1.1 hardware as 0.1.0. The one with the fixes (version and ls) will then get a 0.2.0. Is that OK for you?

peerst, 2017-06-02:
That would be perfect.

commit 32d92d54 (2017-06-02): grisp-bootloader: Add version.

This fixes #8.
closed 2017-06-02 by c-mauderer