grisp/grisp-software #10: GPIO pins including those of SPI1 should be configured safely on startup

peerst, 2017-06-03:
It seems that some GPIO pins are configure to high level output on system startup.

To avoid destroying PMODs that are connected at startup they should be configured
as safe as possible as early as possible (either inputs or totally open if that is even possible)

c-mauderer, 2017-06-06:
The GPIO pins are currently not configured in the grisp/pin-config.h. Therefore they are on default value. I just checked with a debugger for some of them: PD0 to PD3 (which is DIO5 to DIO8 on GPIO2) are configured as input with pull-up enabled by default. That should be a save default configuration.

I'm not aware of a config where the pins are completely disconnected. If no pull-up would be configured, they would be floating. Floating pins are a bad idea too.

If something other then the default is necessary, I need to know which configuration you would prefer. Then I can add it to the pin-config.h and rebuild the bootloader.

peerst, 2017-06-06:
Input with Pullup is fine.  Just was cautions keeping PMODs in their connectors before we know.

So its fine to keep PMODs connected all the time which is the desired behavior

closed 2017-06-06 by peerst