grisp/grisp-software #11: Bootloader complains about entries in grisp.ini file it doesn't know

peerst, 2017-06-04:
Adding a section [network] with a value results in errors in the ini parser of the bootloader e.g.

    boot: error in configuration file: section "network", name "foo", value "bar"

It would be nice if we could use one ini file for bootloader things and application startup things
but these error messages are confusing.

Unknown sections should be ignored, maybe also unknown values in known sections.

c-mauderer, 2017-06-13:
I just had a look at this:
The boot continues, so I would say it's not fatal. But OK, let's remove the messages.

Currently the bootloader knows the `[boot]` section and the `[file]` section. That is because it is based on some other bootloader which did know further options too and the separation was useful. In our case there is only `timeout_in_seconds` and `image_path`. It might could be better to collect these two options in `[boot]`.

If there is only one section for the bootloader we can safely ignore all other sections. A warning would only be reported if there is a unknown key in `[boot]` or if there is no `[boot]` at all. I would rather not ignore unknown keys in `[boot]` to still catch typos.

What do you think?

peerst, 2017-06-13:
Sounds good to me.

commit 2c6e41c1 (2017-07-04): grisp-bootloader: Change config format.

This closes #11.
closed 2017-07-04 by c-mauderer