grisp/grisp-software #15: Under certain conditions the FTDI based console closes repeatedly when Wifi is used

peerst, 2017-07-31:
Might be connected to power issues so far we have seen that behaviour in the following
cases:

* USB2 Hub (self powered) with other devices connected (empty HUB works)

* Unknown Laptop of of a tutorial participant

* Not sure if it also occurred with one of our Macbook Pro laptops

eproxus, 2017-11-06:
Was this ever reproducible?

c-mauderer, 2017-11-06:
No. I could never reproduce it.

peerst, 2017-11-06:
It was occuring on two separate machines at some point but has not been seen in a long while.

I suspect it went away when the Wifi driver got more stable (it only occurred when Wifi was on back then).

So the short answer is: no can't reproduce it anymore.  If anyone can reproduce this problem, please reopen this issue

closed 2017-11-06 by peerst