grisp/grisp-software #17: Error build GDB

eproxus, 2017-10-19:
Trying to build the toolchain from scratch on a new machine, and getting this failure: 
```
shell cmd failed: /bin/sh -ex  /Users/alind/Stritzinger/Code/GRiSP/grisp-software/rtems-source-builder/rtems/build/arm-rtems4.12-gdb-7.12-x86_64-apple-darwin17.0.0-1/doit
error: building arm-rtems4.12-gdb-7.12-x86_64-apple-darwin17.0.0-1
```
(Full report: [rsb-report-arm-rtems4.12-gdb-7.12-x86_64-apple-darwin17.0.0-1.txt](https://github.com/grisp/grisp-software/files/1398159/rsb-report-arm-rtems4.12-gdb-7.12-x86_64-apple-darwin17.0.0-1.txt))

Running that command manually fails with actual errors:

```
/usr/bin/c++ -O2 -pipe -fbracket-depth=1024 -I/Users/alind/Stritzinger/Code/GRiSP/grisp-software/rtems-source-builder/rtems/build/tmp/sb-alind/4.12/rtems-arm/Users/alind/Stritzinger/Code/GRiSP/grisp-software/rtems-install/rtems-4.12/include -g -O2   -I. -I../../gdb-7.12/gdb -I../../gdb-7.12/gdb/common -I../../gdb-7.12/gdb/config -DLOCALEDIR="\"/Users/alind/Stritzinger/Code/GRiSP/grisp-software/rtems-install/rtems-4.12/share/locale\"" -DHAVE_CONFIG_H -I../../gdb-7.12/gdb/../include/opcode -I../../gdb-7.12/gdb/../opcodes/.. -I../../gdb-7.12/gdb/../readline/.. -I../../gdb-7.12/gdb/../zlib -I../bfd -I../../gdb-7.12/gdb/../bfd -I../../gdb-7.12/gdb/../include -I../libdecnumber -I../../gdb-7.12/gdb/../libdecnumber  -I../../gdb-7.12/gdb/gnulib/import -Ibuild-gnulib/import   -DTUI=1  -I/Users/alind/Stritzinger/Code/GRiSP/grisp-software/rtems-source-builder/rtems/build/tmp/sb-alind/4.12/rtems-arm/Users/alind/Stritzinger/Code/GRiSP/grisp-software/rtems-install/rtems-4.12/include  -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -Wall -Wpointer-arith -Wno-unused -Wunused-value -Wunused-function -Wno-switch -Wno-char-subscripts -Wempty-body -Wunused-but-set-parameter -Wunused-but-set-variable -Wno-sign-compare -Wno-write-strings -Wno-narrowing -Wformat-nonliteral  -c -o cp-namespace.o -MT cp-namespace.o -MMD -MP -MF .deps/cp-namespace.Tpo ../../gdb-7.12/gdb/cp-namespace.c
../../gdb-7.12/gdb/doublest.c:258:19: error: use of undeclared identifier 'min'; did you mean 'fmin'?
      mant_bits = min (mant_bits_left, 32);
                  ^~~
                  fmin
/Library/Developer/CommandLineTools/usr/include/c++/v1/math.h:1193:1: note: 'fmin' declared here
fmin(_A1 __lcpp_x, _A2 __lcpp_y) _NOEXCEPT
^
In file included from ../../gdb-7.12/gdb/frame-base.c:20:
In file included from ../../gdb-7.12/gdb/defs.h:28:
../../gdb-7.12/gdb/common/common-defs.h:47:9: warning: '__STDC_CONSTANT_MACROS' macro redefined [-Wmacro-redefined]
#define __STDC_CONSTANT_MACROS 1
        ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/stdint.h:116:12: note: previous definition is here
#   define __STDC_CONSTANT_MACROS
           ^
In file included from ../../gdb-7.12/gdb/frame-base.c:20:
In file included from ../../gdb-7.12/gdb/defs.h:28:
../../gdb-7.12/gdb/common/common-defs.h:48:9: warning: '__STDC_LIMIT_MACROS' macro redefined [-Wmacro-redefined]
#define __STDC_LIMIT_MACROS 1
        ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/stdint.h:113:12: note: previous definition is here
#   define __STDC_LIMIT_MACROS
           ^
../../gdb-7.12/gdb/doublest.c:568:19: error: use of undeclared identifier 'min'; did you mean 'fmin'?
      mant_bits = min (mant_bits_left, 32);
                  ^~~
                  fmin
/Library/Developer/CommandLineTools/usr/include/c++/v1/math.h:1193:1: note: 'fmin' declared here
fmin(_A1 __lcpp_x, _A2 __lcpp_y) _NOEXCEPT
^
../../gdb-7.12/gdb/doublest.c:912:25: error: use of undeclared identifier 'min'; did you mean 'fmin'?
      memcpy (to, from, min (TYPE_LENGTH (from_type), TYPE_LENGTH (to_type)));
                        ^~~
                        fmin
/Library/Developer/CommandLineTools/usr/include/c++/v1/math.h:1193:1: note: 'fmin' declared here
fmin(_A1 __lcpp_x, _A2 __lcpp_y) _NOEXCEPT
^
6 warnings and 3 errors generated.
```

(Full log: [gdb-doit.log](https://github.com/grisp/grisp-software/files/1398188/gdb-doit.log))

Any suggestions?

eproxus, 2017-10-19:
Seems that version of GDB is not compatible with macOS 10.13 High Sierra 😞 

https://trac.macports.org/ticket/55035

eproxus, 2017-10-19:
Possibly related: https://sourceware.org/bugzilla/show_bug.cgi?id=20863

c-mauderer, 2017-10-20:
Do you know whether that problem is only with a special version of gdb or with all versions?
Do you need gdb? Otherwise it might could be possible to find an option for the RSB to not build gdb. Something like `--without-gdb` but that is just a guess.

eproxus, 2017-10-20:
Workaround for now is to comment out GDB in `rtems-source-builder/rtems/config/4.12/rtems-default.bset`.

eproxus, 2017-11-06:
I'm setting this to the 1.0 milestone because it should either be fixed, or at least documented how to workaround the issue.

peerst, 2017-11-14:
I think this is getting more necessary now.  If we and our users can't build gdb we can't JTAG debug with Open OCD neither if I understand this correctly?

c-mauderer, 2017-11-15:
OpenOCD can do very few on it's own mostly because (as far as I know) it can't process any ELF files. But it can for example update the flash without any gdb. So without gdb there are only very basic functions, yes.

Regarding the bug: As far as I understand it, this is an upstream bug of gdb? Do you know whether it is already fixed in a newer version? If yes, we could update the toolchain and RTEMS versions to the RTEMS 5 release branch. The RTEMS 5 branch uses an gdb 8.0.1.

I can't really try that out here because I don't have any macOS computers. So I'll have to rely mostly on your feedback.

peerst, 2017-11-15:
IIRC the gdb problem was only in the simulators so if I build a normal gdb it will have no problem.

Question is if we want to open the can of worms of bumping the toolchain before a 1.0 version.

Maybe we should just disable gdb by default in the build script?  Then we have time to upgrade the toolchains and test them for a 1.1 in a few weeks

Or maybe we could just disable the simulator building (after verifying its in the simulator part)?

c-mauderer, 2017-11-15:
Basically you have to decide which way to go. All options should be possible.

If we just disable the gdb build, of course your users (and that means _all_ users and not only the ones using a macOS) won't be able to debug too.

Note that for the later two options (disabling gdb or simulator) we might have to fork the rtems-source-builder. That's not really a problem but it means that we are on a custom branch there too for the release.

Again: I can't really test the build due to the lack of a build host with macOS. So if you want I can prepare it on some branch and push it for testing.

peerst, 2017-11-15:
Good point about the problem only occuring on Mac OS.

Not building gdb on Mac OS is only a cosmetic solution since the result is the same it just 
doesn't cause the error.  So forking source builder for just not building gdb is not really worth it.

Is there a flag like --without-gdb which we could specify conditionally from our build script for Mac OS?

We could also output a explanatory message at the end of the script like "If you are on a Mac and the last error was about building gdb this is currently normal"

c-mauderer, 2017-11-15:
I just checked it: RSB currently doesn't have a `--without-gdb` option. But the simulator is already disabled on some system / target combinations. So disabling it in a fork should not be too hard.

peerst, 2017-11-15:
Ok, then lets fork the source builder.  @c-mauderer could you do this in a branch on the RSB repo or do we need to fork it here?  Might be interesting for other Mac users of RTEMS too

c-mauderer, 2017-11-15:
I don't have access to the upstream RSB repo so I would fork it here. If it works, we can send a patch to the mailing list to get this fix upstream too.

I added a branch called 20171115-gdb-macOS to the grisp-software repository. That one contains a RSB version which should disable the simulator in the gdb-7.12 on systems matching "darwin". I also cherry-picked the xz-patch for fixing #3.

Note that you need the following commands when changing to that branch:
```
git checkout --track origin/20171115-gdb-macOS
git submodule sync
git submodule update
```
The sync is necessary to update the URL of the submodule.

Could you try whether the gdb can now be build?

peerst, 2017-11-15:
Hi Christian,

built from scratch after cloning&checkout … however the problem persists.

Something with “min” not being declared.

package: arm-rtems4.12-gdb-7.12-x86_64-apple-darwin16.7.0-1
download: http://ftp.gnu.org/gnu/gdb/gdb-7.12.tar.xz -> sources/gdb-7.12.tar.xz
downloading: sources/gdb-7.12.tar.xz - 18.3MB of 18.3MB (100%)  
download: https://gaisler.org/gdb/gdb-7.12-sis-leon2-leon3.diff -> patches/gdb-7.12-sis-leon2-leon3.diff
downloading: patches/gdb-7.12-sis-leon2-leon3.diff - 165.0kB of 165.0kB (100%) 
download: https://sourceware.org/git/gitweb.cgi?p=binutils-gdb.git;a=blobdiff_plain;f=gdb/location.c;h=8dce21ada12b3f8cb4145d1af96e29efe378bda2;hp=65116c732f7071444986c4b582580b7cc4622f0b;hb=909de...<see log> -> patches/binutils-gdb-git-8dce21ada12b3f8cb4145d1af96e29efe378bda2.patch
downloading: patches/binutils-gdb-git-8dce21ada12b3f8cb4145d1af96e29efe378bda2.patch - 743.0 bytes 
download: https://sourceware.org/cgi-bin/get-raw-msg?listname=gdb-patches&date=2016-11&msgid=20161124152917.52189-1-jhb%40FreeBSD.org -> patches/PATCH_Define_an_error_function_in_the_PPC_simulator
downloading: patches/PATCH_Define_an_error_function_in_the_PPC_simulator - 3.0kB     
building: arm-rtems4.12-gdb-7.12-x86_64-apple-darwin16.7.0-1
error: building arm-rtems4.12-gdb-7.12-x86_64-apple-darwin16.7.0-1
Build FAILED
  See error report: rsb-report-arm-rtems4.12-gdb-7.12-x86_64-apple-darwin16.7.0-1.txt
error: building arm-rtems4.12-gdb-7.12-x86_64-apple-darwin16.7.0-1
Build Set: Time 0:36:47.758038
Build FAILED

> I don't have access to the upstream RSB repo so I would fork it here. If it works, we can send a patch to the mailing list to get this fix upstream too.
>
> I added a branch called 20171115-gdb-macOS [...]

RTEMS Tools Project - Source Builder Error Report
 Build: error: building arm-rtems4.12-gdb-7.12-x86_64-apple-darwin16.7.0-1
 Command Line: ../source-builder/sb-set-builder --log=/Users/peer/projects/grisp/grisp-software/build/..//build//rsb-arm-20171115_132714.log --prefix=/Users/peer/projects/grisp/grisp-software/build/..//rtems-install/rtems-4.12/ --no-clean --without-rtems 4.12/rtems-arm
 Python: 2.7.10 (default, Feb  7 2017, 00:08:15) [GCC 4.2.1 Compatible Apple LLVM 8.0.0 (clang-800.0.34)]
 https://github.com/grisp/rtems-source-builder/origin/4c8b33755b7bc6d86f1ddfd8c311e1753b2a4150-modified
 Darwin yrael.lan 16.7.0 Darwin Kernel Version 16.7.0: Thu Jun 15 17:36:27 PDT 2017; root:xnu-3789.70.16~2/RELEASE_X86_64 x86_64
Tail of the build log:
#   define __STDC_CONSTANT_MACROS
           ^
In file included from ../../gdb-7.12/gdb/top.c:20:
In file included from ../../gdb-7.12/gdb/defs.h:28:
../../gdb-7.12/gdb/common/common-defs.h:48:9: warning: '__STDC_LIMIT_MACROS' macro redefined [-Wmacro-redefined]
#define __STDC_LIMIT_MACROS 1
        ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/stdint.h:113:12: note: previous definition is here
#   define __STDC_LIMIT_MACROS
           ^
/usr/bin/c++ -O2 -pipe -fbracket-depth=1024 -I/Users/peer/projects/grisp/grisp-software/rtems-source-builder/rtems/build/tmp/sb-peer/4.12/rtems-arm/Users/peer/projects/grisp/grisp-software/rtems-install/rtems-4.12/include -g -O2   -I. -I../../gdb-7.12/gdb -I../../gdb-7.12/gdb/common -I../../gdb-7.12/gdb/config -DLOCALEDIR="\"/Users/peer/projects/grisp/grisp-software/rtems-install/rtems-4.12/share/locale\"" -DHAVE_CONFIG_H -I../../gdb-7.12/gdb/../include/opcode -I../../gdb-7.12/gdb/../opcodes/.. -I../../gdb-7.12/gdb/../readline/.. -I../../gdb-7.12/gdb/../zlib -I../bfd -I../../gdb-7.12/gdb/../bfd -I../../gdb-7.12/gdb/../include -I../libdecnumber -I../../gdb-7.12/gdb/../libdecnumber  -I../../gdb-7.12/gdb/gnulib/import -Ibuild-gnulib/import   -DTUI=1  -I/Users/peer/projects/grisp/grisp-software/rtems-source-builder/rtems/build/tmp/sb-peer/4.12/rtems-arm/Users/peer/projects/grisp/grisp-software/rtems-install/rtems-4.12/include  -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -Wall -Wpointer-arith -Wno-unused -Wunused-value -Wunused-function -Wno-switch -Wno-char-subscripts -Wempty-body -Wunused-but-set-parameter -Wunused-but-set-variable -Wno-sign-compare -Wno-write-strings -Wno-narrowing -Wformat-nonliteral  -c -o utils.o -MT utils.o -MMD -MP -MF .deps/utils.Tpo ../../gdb-7.12/gdb/utils.c
/usr/bin/c++ -O2 -pipe -fbracket-depth=1024 -I/Users/peer/projects/grisp/grisp-software/rtems-source-builder/rtems/build/tmp/sb-peer/4.12/rtems-arm/Users/peer/projects/grisp/grisp-software/rtems-install/rtems-4.12/include -g -O2   -I. -I../../gdb-7.12/gdb -I../../gdb-7.12/gdb/common -I../../gdb-7.12/gdb/config -DLOCALEDIR="\"/Users/peer/projects/grisp/grisp-software/rtems-install/rtems-4.12/share/locale\"" -DHAVE_CONFIG_H -I../../gdb-7.12/gdb/../include/opcode -I../../gdb-7.12/gdb/../opcodes/.. -I../../gdb-7.12/gdb/../readline/.. -I../../gdb-7.12/gdb/../zlib -I../bfd -I../../gdb-7.12/gdb/../bfd -I../../gdb-7.12/gdb/../include -I../libdecnumber -I../../gdb-7.12/gdb/../libdecnumber  -I../../gdb-7.12/gdb/gnulib/import -Ibuild-gnulib/import   -DTUI=1  -I/Users/peer/projects/grisp/grisp-software/rtems-source-builder/rtems/build/tmp/sb-peer/4.12/rtems-arm/Users/peer/projects/grisp/grisp-software/rtems-install/rtems-4.12/include  -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -Wall - [... cut ...]
make[2]: *** [doublest.o] Error 1
make[2]: *** Waiting for unfinished jobs....
4 warnings generated.
4 warnings generated.
4 warnings generated.
4 warnings generated.
4 warnings generated.
4 warnings generated.
101 warnings generated.
make[1]: *** [all-gdb] Error 2
make: *** [all] Error 2
shell cmd failed: /bin/sh -ex  /Users/peer/projects/grisp/grisp-software/rtems-source-builder/rtems/build/arm-rtems4.12-gdb-7.12-x86_64-apple-darwin16.7.0-1/doit
error: building arm-rtems4.12-gdb-7.12-x86_64-apple-darwin16.7.0-1

peerst, 2017-11-15:
If I understand correctly it builds a C file with the c++ compiler and that resulted in this error

c-mauderer, 2017-11-15:
Hm, `gdb/doublest.c` contains `/* Floating point routines for GDB, the GNU debugger. ... */`. That doesn't sound like something simulator related but more a general one. So I think there are only a few options:

1. Simplest one is to just disable the gdb build on the branch instead of only the simulator.

2. One which would need a lot more time and testing would be to find whether the problem has been already fixed upstream and backport the patches. Most likely it would be the same as on FreeBSD: https://bugs.freebsd.org/bugzilla//show_bug.cgi?id=219126

I would expect that the patches are included in the gdb 8 release. So if you plan to update the tool chain for the 1.1 release we won't have the problem there. So solution two would be a lot of effort for fixing a problem in a release that will most likely soon be superseded. Therefore I would suggest to go for solution 1 for now. What do you think?

peerst, 2017-11-15:
Lets disable building gdb for now and update to the new tools for 1.1

nextl00p, 2017-11-15:
I don't know if there is any relationship between this and my observings:

On a fresh Fedora 27 installation gdb also failed to build. I had to install headers for python2 (python2-devel), then it worked.

c-mauderer, 2017-11-16:
@peerst: I updated the branch with a patch that should disable the gdb on darwin. I'm not 100% sure whether the `%if %{_build_os} != darwin` if works at that point. So could you please try it?

@nextl00p: No, that problem isn't related. Your problem would be related to #5 and #7.

eproxus, 2017-11-27:
@c-mauderer If I recall correctly from some cursory web searching, it could potentially be fixed in 8.0.? (don't remember the exact version). My knowledge of GCC is basically non-existent though so someone with more experience should try to figure it out 😄

c-mauderer, 2017-11-27:
Hello @eproxus,

that would match my recherche from the comment a little earlier https://github.com/grisp/grisp-software/issues/17#issuecomment-344592580. @peerst decided that the tool chain upgrade should be in 1.1. So disabling the gdb would be a workaround for building the 1.0 tool chain. I only never got a response whether the workaround works. Therefore it hasn't been merged yet.

Regards

Christian

eproxus, 2017-11-27:
@c-mauderer Ah, missed that, sorry for the noise.

I'll try a rebuild today with the branch and report back

eproxus, 2017-11-27:
@c-mauderer Clean build of branch `20171115-gdb-macOS` fails:

```
» ./build/build.sh                                                         ⌗1 2s
+ set -e -u
+++ dirname -- ./build/build.sh
++ CDPATH=
++ cd -- ./build
++ pwd
+ SCRIPTDIR=/Users/alind/Stritzinger/Code/GRiSP/grisp-software/build
+ PROJECTDIR=/Users/alind/Stritzinger/Code/GRiSP/grisp-software/build/../
+ source /Users/alind/Stritzinger/Code/GRiSP/grisp-software/build/configuration.sh
++ BSP_NAME=atsamv
++ RTEMS_CPU=arm
++ RTEMS_VERSION=4.12
++ TARGET=arm-rtems4.12
++ PREFIX=/Users/alind/Stritzinger/Code/GRiSP/grisp-software/build/..//rtems-install/rtems-4.12/
++ RTEMS_SOURCE_DIR=/Users/alind/Stritzinger/Code/GRiSP/grisp-software/build/..//rtems
++ LIBBSD_SOURCE_DIR=/Users/alind/Stritzinger/Code/GRiSP/grisp-software/build/..//rtems-libbsd
++ LIBINIH_SOURCE_DIR=/Users/alind/Stritzinger/Code/GRiSP/grisp-software/build/..//libinih
++ LIBGRISP_SOURCE_DIR=/Users/alind/Stritzinger/Code/GRiSP/grisp-software/build/..//libgrisp
++ BSP_CONFIG_OPT=("--disable-tests" "--disable-networking" "--enable-chip=same70q21" "--enable-sdram=is42s16320f-7bl" "ATSAM_CONSOLE_DEVICE_TYPE=1" "ATSAM_CONSOLE_DEVICE_INDEX=2" "ATSAM_MEMORY_QSPIFLASH_SIZE=0x0" "ATSAM_MEMORY_NOCACHE_SIZE=0x8000")
+ BUILD_DIR=/Users/alind/Stritzinger/Code/GRiSP/grisp-software/build/..//build/b-atsamv
+ export 'PATH=/Users/alind/Stritzinger/Code/GRiSP/grisp-software/build/..//rtems-install/rtems-4.12//bin:/Users/alind/.asdf/bin:/Users/alind/.asdf/shims:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Applications/VMware Fusion.app/Contents/Public:/usr/local/MacGPG2/bin:/Applications/Wireshark.app/Contents/MacOS:/Users/alind/.bin:/usr/local/sbin:/Users/alind/.asdf/bin:/Users/alind/.asdf/shims:/Users/alind/Code/go/bin:/Users/alind/.psvm/current/bin'
+ PATH='/Users/alind/Stritzinger/Code/GRiSP/grisp-software/build/..//rtems-install/rtems-4.12//bin:/Users/alind/.asdf/bin:/Users/alind/.asdf/shims:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Applications/VMware Fusion.app/Contents/Public:/usr/local/MacGPG2/bin:/Applications/Wireshark.app/Contents/MacOS:/Users/alind/.bin:/usr/local/sbin:/Users/alind/.asdf/bin:/Users/alind/.asdf/shims:/Users/alind/Code/go/bin:/Users/alind/.psvm/current/bin'
+ DO_SUBMODULES=1
+ DO_TOOLCHAIN=1
+ DO_BOOTSTRAP=1
+ '[' 1 -ne 0 ']'
+ cd /Users/alind/Stritzinger/Code/GRiSP/grisp-software/build/../
+ git submodule init
+ git submodule update
error: Server does not allow request for unadvertised object ba516308ba1989042cb4e5ffe36d1ec0d41a9d50
Fetched in submodule path 'rtems-source-builder', but it did not contain ba516308ba1989042cb4e5ffe36d1ec0d41a9d50. Direct fetching of that commit failed.
```

c-mauderer, 2017-11-27:
Clean build or fresh clone?

If you didn't do a fresh clone, the submodule path would be wrong. That can lead to the given message. I added a `git submodule sync` to the build script just now to catch that. Could you please re-try?

eproxus, 2017-11-27:
🤦‍♂️ that was it... checkout and cleaned with `git clean -fxd && git clean -fXd` and now building...

c-mauderer, 2017-11-28:
If this still fails, I think we have three possibilities:
- Completely disable gdb for _all_ systems.
- Document that as a known bug for 1.0 and just say that it will be fixed with the new toolchain in 1.1.
- Put a bigger amount of work into understanding why that work around didn't work. But I don't really think that is a good idea if a toolchain update to 1.1 will fix it anyway.

eproxus, 2017-11-28:
Agree, I think document and move to 1.1 as soon as possible is the best way forward

peerst, 2018-03-26:
Actually gdb building should have been disabled by @c-mauderer for darwin according to comment on Nov 16th 2017.

So why do new users still get gdb build errors?

This patch should be urgently tested and merged into whatever version we let our users download immediately if it works

eproxus, 2018-03-26:
The branch 20171115-gdb-macOS does _not_ work, GDB is built regardless.

c-mauderer, 2018-03-27:
Back then I interpreted the

> Agree, I think document and move to 1.1 as soon as possible is the best way forward

that you wanted to move to a new tool chain soon. From my point of view that would still be the best solution. So I would still suggest this.

Your RTEMS version is still based on the one somewhere around middle of 2017. Back then the tool chain and newlib version has still been quite unstable (if I remember correctly due to support for some new C++ standard). But since a few months, the version has been quite stable. RTEMS 5 now uses release versions:
 * Binutils 2.30
 * Newlib 3.0.0
 * GCC 7.3.0

Disabling gdb only for Mac OS didn't work as expected. I can't really test that because I don't have a Mac. Doing that hack by guessing and sending it to one of you for testing isn't really efficient. Therefore the next best alternative that I can offer would be to disable gdb for all systems. I don't really think that this is a good solution but if you like, I can do that too.

nextl00p, 2018-06-05:
I just could build this with RTEMS 5 on mac osx 10.13.4 without issues.

closed 2018-06-05 by nextl00p