grisp/grisp-software #20: Everything in ./build should be Bash scripts not shell scripts

nextl00p, 2017-11-15:
I just tried to setup the buildchain in Ubuntu 16.04. When running `./build/build.sh' it complains:
```
source /home/kilian/grisp-software/build/configuration.sh
./build/build.sh: 13: ./build/build.sh: source: not found
```
Whereas if I run with `bash ./build/build.sh` it starts off nice, but fails at a later point, where it runs again into a similar error.

So i think all hashbangs should be changed to `#! /usr/bin/env bash`.

See also: https://stackoverflow.com/questions/670191/getting-a-source-not-found-error-when-using-source-in-a-bash-script

In Fedora  (also an many other OSs) `sh` defaults to bash.
In Ubuntu `sh` defaults to dash.

c-mauderer, 2017-11-15:
My intention was to write the scripts POSIX compliant. So any shell that respects POSIX should be able to execute it. It seems that I missed that `source` is not POSIX. But the stackoverflow thread you linked already contains the correct fix: Replacing `source` by a `.`: https://stackoverflow.com/a/670209

I'll create a fix for that.

peerst, 2017-11-15:
Yeah POSIX shell compliance is topmost importance.

I don’t want to get into a bash dependency here without necessity so I agree to move towards `.`

Didn’t know that `source` is not POSIX compliant either

commit 934cf9a2 (2017-11-16): build: Improve POSIX compliance of scripts.

Now the build scripts should work with dash too (used as sh for example
on Debian and Ubuntu).

This fixes #20.
closed 2017-11-16 by c-mauderer