grisp/grisp-software #21: Building fails under Ubuntu 16.04

nextl00p, 2017-11-16:
`./build/build.sh' fails with some weird linker error quite late in the process.

```
[2159/2208] Linking build/arm-rtems4.12-atsamv/arphole.exe
[2160/2208] Linking build/arm-rtems4.12-atsamv/cdev01.exe
[2161/2208] Linking build/arm-rtems4.12-atsamv/commands01.exe
[2162/2208] Linking build/arm-rtems4.12-atsamv/condvar01.exe
[2163/2208] Linking build/arm-rtems4.12-atsamv/debugger01.exe
[2164/2208] Linking build/arm-rtems4.12-atsamv/dhcpcd01.exe
linkcmds.base:392 cannot move location counter backwards (from 0000000020480f38 to 0000000020458000)
collect2: error: ld returned 1 exit status
linkcmds.base:392 cannot move location counter backwards (from 00000000205224a0 to 0000000020458000)
collect2: error: ld returned 1 exit status
linkcmds.base:392 cannot move location counter backwards (from 00000000205237e0 to 0000000020458000)
collect2: error: ld returned 1 exit status
Waf: Leaving directory `/home/kilian/grisp-software/rtems-libbsd/build/arm-rtems4.12-atsamv'
Build failed
 -> task in 'commands01.exe' failed with exit status 1 (run with -v to display more information)
 -> task in 'debugger01.exe' failed with exit status 1 (run with -v to display more information)
 -> task in 'dhcpcd01.exe' failed with exit status 1 (run with -v to display more information)
```

I use a Ubuntu 16.04 VM with

`sudo apt-get install git zlib1g-dev bison flex texinfo curl automake autoconf libreadline-dev libncurses-dev libssl-dev libyaml-dev libxslt-dev libffi-dev libtool unixodbc-dev python2.7-dev`

and

   ```
   $ mkdir ~/bin 
   $ curl -o ~/bin/waf https://waf.io/waf-2.0.2
   $ chmod +x ~/bin/waf
   ```

c-mauderer, 2017-11-17:
Basically the linker error tells you that there is too much code for the application and it doesn't fit in the available memory. For the libbsd examples, that happened if the applications are built so that they are executed from the internal flash instead of the SDRAM. To avoid that the `build-libbsd.sh` implements a workaround that uses the linkcmds for the SDRAM. It seems the workarround didn't work here.

Do you have the log before the libbsd is build. There should be a lot of lines starting with `+` and `++`. That is the output of the shell script.

c-mauderer, 2017-11-17:
I tried it with dash on OpenSUSE and had no problem. But I'm not 100% sure whether it is enough to just call `dash ./build/build.sh` to replace every sh call with a dash. So the sub-scripts might still have used bash.

But I got the relevant log output. Shortly before libbsd is build, you should get that one. Please take a special note whether the four bold lines are the same (except for the path and that they won't appear bold in your log).
<pre>
[...]
+ /home/EB/christian_m/Projekte/grisp-software-test/build/build-libbsd.sh clean install
+ set -e -u
+++ dirname -- /home/EB/christian_m/Projekte/grisp-software-test/build/build-libbsd.sh
++ CDPATH=
++ cd -- /home/EB/christian_m/Projekte/grisp-software-test/build
++ pwd
+ SCRIPTDIR=/home/EB/christian_m/Projekte/grisp-software-test/build
+ PROJECTDIR=/home/EB/christian_m/Projekte/grisp-software-test/build/../
+ . /home/EB/christian_m/Projekte/grisp-software-test/build/configuration.sh
++ BSP_NAME=atsamv
++ RTEMS_CPU=arm
++ RTEMS_VERSION=4.12
++ TARGET=arm-rtems4.12
++ PREFIX=/home/EB/christian_m/Projekte/grisp-software-test/build/..//rtems-install/rtems-4.12/
++ RTEMS_SOURCE_DIR=/home/EB/christian_m/Projekte/grisp-software-test/build/..//rtems
++ LIBBSD_SOURCE_DIR=/home/EB/christian_m/Projekte/grisp-software-test/build/..//rtems-libbsd
++ LIBINIH_SOURCE_DIR=/home/EB/christian_m/Projekte/grisp-software-test/build/..//libinih
++ LIBGRISP_SOURCE_DIR=/home/EB/christian_m/Projekte/grisp-software-test/build/..//libgrisp
++ BSP_CONFIG_OPT='
        --disable-tests
        --disable-networking
        --enable-chip=same70q21
        --enable-sdram=is42s16320f-7bl
        ATSAM_CONSOLE_DEVICE_TYPE=1
        ATSAM_CONSOLE_DEVICE_INDEX=2
        ATSAM_MEMORY_QSPIFLASH_SIZE=0x0
        ATSAM_MEMORY_NOCACHE_SIZE=0x8000
        '
+ export PATH=/home/EB/christian_m/Projekte/grisp-software-test/build/..//rtems-install/rtems-4.12//bin:/home/EB/christian_m/Projekte/grisp-software-test/build/..//rtems-install/rtems-4.12//bin:/home/EB/christian_m/perl5/bin:/home/EB/christian_m/perl5/bin:/usr/local/bin:/usr/bin:/bin:/usr/bin/X11:/usr/games:/usr/lib/mit/bin:/sbin:/usr/sbin:/home/EB/christian_m/scripts:/home/EB/christian_m/t32/bin/pc_linux64:/sbin:/usr/sbin:/home/EB/christian_m/scripts:/home/EB/christian_m/t32/bin/pc_linux64
+ PATH=/home/EB/christian_m/Projekte/grisp-software-test/build/..//rtems-install/rtems-4.12//bin:/home/EB/christian_m/Projekte/grisp-software-test/build/..//rtems-install/rtems-4.12//bin:/home/EB/christian_m/perl5/bin:/home/EB/christian_m/perl5/bin:/usr/local/bin:/usr/bin:/bin:/usr/bin/X11:/usr/games:/usr/lib/mit/bin:/sbin:/usr/sbin:/home/EB/christian_m/scripts:/home/EB/christian_m/t32/bin/pc_linux64:/sbin:/usr/sbin:/home/EB/christian_m/scripts:/home/EB/christian_m/t32/bin/pc_linux64
+ cd /home/EB/christian_m/Projekte/grisp-software-test/build/..//rtems-libbsd
<b>+ '[' atsamv == atsamv ']'
+ '[' '!' -e /home/EB/christian_m/Projekte/grisp-software-test/build/..//rtems-install/rtems-4.12//arm-rtems4.12/atsamv/lib/linkcmds.org ']'
+ mv /home/EB/christian_m/Projekte/grisp-software-test/build/..//rtems-install/rtems-4.12//arm-rtems4.12/atsamv/lib/linkcmds /home/EB/christian_m/Projekte/grisp-software-test/build/..//rtems-install/rtems-4.12//arm-rtems4.12/atsamv/lib/linkcmds.org
+ cp /home/EB/christian_m/Projekte/grisp-software-test/build/..//rtems-install/rtems-4.12//arm-rtems4.12/atsamv/lib/linkcmds.sdram /home/EB/christian_m/Projekte/grisp-software-test/build/..//rtems-install/rtems-4.12//arm-rtems4.12/atsamv/lib/linkcmds</b>
+ DO_CLEAN=0
+ DO_INSTALL=0
+ for i in '"$@"'
+ case "$i" in
+ DO_CLEAN=1
+ for i in '"$@"'
+ case "$i" in
+ DO_INSTALL=1
+ '[' 1 -ne 0 ']'
+ '[' -e build ']'
+ waf configure --prefix=/home/EB/christian_m/Projekte/grisp-software-test/build/..//rtems-install/rtems-4.12/ --rtems-bsps=arm/atsamv
Setting top to                           : /home/EB/christian_m/Projekte/grisp-software-test/rtems-libbsd 
Setting out to                           : /home/EB/christian_m/Projekte/grisp-software-test/rtems-libbsd/build 
RTEMS Version                            : 4.12 
Architectures                            : arm-rtems4.12 
Board Support Package                    : arm-rtems4.12-atsamv 
Show commands                            : no 
Long commands                            : no 
[...]
</pre>

nextl00p, 2017-11-17:
Seems to be a Dash/Bash error.

See the entire [build.log](https://github.com/grisp/grisp-software/files/1481927/build.log).

Relevant part:

```
+ BSP_NAME=atsamv
+ RTEMS_CPU=arm
+ RTEMS_VERSION=4.12
+ TARGET=arm-rtems4.12
+ PREFIX=/home/kilian/grisp-software/build/..//rtems-install/rtems-4.12/
+ RTEMS_SOURCE_DIR=/home/kilian/grisp-software/build/..//rtems
+ LIBBSD_SOURCE_DIR=/home/kilian/grisp-software/build/..//rtems-libbsd
+ LIBINIH_SOURCE_DIR=/home/kilian/grisp-software/build/..//libinih
+ LIBGRISP_SOURCE_DIR=/home/kilian/grisp-software/build/..//libgrisp
+ BSP_CONFIG_OPT=
	--disable-tests
	--disable-networking
	--enable-chip=same70q21
	--enable-sdram=is42s16320f-7bl
	ATSAM_CONSOLE_DEVICE_TYPE=1
	ATSAM_CONSOLE_DEVICE_INDEX=2
	ATSAM_MEMORY_QSPIFLASH_SIZE=0x0
	ATSAM_MEMORY_NOCACHE_SIZE=0x8000
	
+ export PATH=/home/kilian/grisp-software/build/..//rtems-install/rtems-4.12//bin:/home/kilian/grisp-software/build/..//rtems-install/rtems-4.12//bin:/home/kilian/bin:/home/kilian/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
+ cd /home/kilian/grisp-software/build/..//rtems-libbsd
+ [ atsamv == atsamv ]
/home/kilian/grisp-software/build/build-libbsd.sh: 18: [: atsamv: unexpected operator
+ DO_CLEAN=0
+ DO_INSTALL=0
+ DO_CLEAN=1
+ DO_INSTALL=1
+ [ 1 -ne 0 ]
+ [ -e build ]
+ waf clean
'clean-arm-rtems4.12-atsamv' finished successfully (0.410s)
+ waf configure --prefix=/home/kilian/grisp-software/build/..//rtems-install/rtems-4.12/ --rtems-bsps=arm/atsamv
Setting top to                           : /home/kilian/grisp-software/rtems-libbsd 
Setting out to                           : /home/kilian/grisp-software/rtems-libbsd/build 
RTEMS Version                            : 4.12 
Architectures                            : arm-rtems4.12 
Board Support Package                    : arm-rtems4.12-atsamv 
Show commands                            : no 
Long commands                            : no 
```

c-mauderer, 2017-11-17:
Yes. Seems that two equal signs are not POSIX but work in bash. Should be fixed now.

commit a8943b6d (2017-11-17): build: Improve POSIX compliance.

This should fix #21.
closed 2017-11-17 by c-mauderer