grisp/grisp-software #24: Build fails on Mac OS 10.10.5

doppioslash, 2017-11-22:
Following [instructions](https://github.com/grisp/grisp/wiki/macOS-Development-Environment), this happens, after calling `./build/build.sh`:

```
Waf: Leaving directory `/[...]/grisp-software/rtems-libbsd/build/arm-rtems4.12-atsamv'
'install-arm-rtems4.12-atsamv' finished successfully (3.752s)
+ '[' atsamv == atsamv ']'
+ mv /[...]/grisp-software/build/..//rtems-install/rtems-4.12//arm-rtems4.12/atsamv/lib/linkcmds.org /[...]/grisp-software/build/..//rtems-install/rtems-4.12//arm-rtems4.12/atsamv/lib/linkcmds
+ /[...]/grisp-software/build/build-libinih.sh
+ set -e -u
+++ dirname -- /[...]/grisp-software/build/build-libinih.sh
++ CDPATH=
++ cd -- /[...]/grisp-software/build
++ pwd
+ SCRIPTDIR=/[...]/grisp-software/build
+ PROJECTDIR=/[...]/grisp-software/build/../
+ source /[...]/grisp-software/build/configuration.sh
++ BSP_NAME=atsamv
++ RTEMS_CPU=arm
++ RTEMS_VERSION=4.12
++ TARGET=arm-rtems4.12
++ PREFIX=/[...]/grisp-software/build/..//rtems-install/rtems-4.12/
++ RTEMS_SOURCE_DIR=/[...]/grisp-software/build/..//rtems
++ LIBBSD_SOURCE_DIR=/[...]/grisp-software/build/..//rtems-libbsd
++ LIBINIH_SOURCE_DIR=/[...]/grisp-software/build/..//libinih
++ LIBGRISP_SOURCE_DIR=/[...]/grisp-software/build/..//libgrisp
++ BSP_CONFIG_OPT=("--disable-tests" "--disable-networking" "--enable-chip=same70q21" "--enable-sdram=is42s16320f-7bl" "ATSAM_CONSOLE_DEVICE_TYPE=1" "ATSAM_CONSOLE_DEVICE_INDEX=2" "ATSAM_MEMORY_QSPIFLASH_SIZE=0x0" "ATSAM_MEMORY_NOCACHE_SIZE=0x8000")
+ export 'PATH=/[...]/grisp-software/build/..//rtems-install/rtems-4.12//bin:[...]/grisp-software/build/..//rtems-install/rtems-4.12//bin:/usr/local/opt/asdf/bin:/usr/local/opt/asdf/shims
+ PATH='/[...]/grisp-software/build/..//rtems-install/rtems-4.12//bin:/[...]/grisp-software/build/..//rtems-install/rtems-4.12//bin:/usr/local/opt/asdf/bin:/usr/local/opt/asdf/shims
+ cd /[...]/grisp-software/build/..//libinih
+ make clean install
rm -rf b-atsamv
mkdir b-atsamv
mkdir b-atsamv/inih
arm-rtems4.12-ar rcu b-atsamv/libinih.a
arm-rtems4.12-ar: `u' modifier ignored since `D' is the default (see `U')
arm-rtems4.12-ranlib b-atsamv/libinih.a
mkdir -p /[...]/grisp-software/libinih/../rtems-install/rtems-4.12/arm-rtems4.12/atsamv/lib/include/inih
install -m 644 b-atsamv/libinih.a /[...]/grisp-software/libinih/../rtems-install/rtems-4.12/arm-rtems4.12/atsamv/lib
install -m 644 inih/*.h /[...]/grisp-software/libinih/../rtems-install/rtems-4.12/arm-rtems4.12/atsamv/lib/include/inih
install: inih/*.h: No such file or directory
make: *** [install] Error 71
```

c-mauderer, 2017-11-23:
From the log I would say that the libini/inih submodule isn't initialized. Could you check the following:
- Does `git submodule status` write a "`-`" before the commit id of inih?
- Does `git status` print something odd?

doppioslash, 2017-11-27:
@c-mauderer here's the output of `git submodule status`, `git status` and `git diff`:
```
$ git submodule status
 5dbf5cb6b4027d5937726b8c499bd93c5b7d935d libinih/inih (r36)
 e5d4a895e23797130b82c90fede0cc15586d9fce rtems (e5d4a895e2)
 1f9ac0d707cbb3cf40b344c01f5f6e0a6cce6b07 rtems-libbsd (remotes/origin/grisp-alpha-20161116-179-g1f9ac0d7)
 1ceafe5f12744e4b4424e68c70e6a5379c481e7e rtems-source-builder (1ceafe5)
$ git status
HEAD detached at grisp_base-1.2-beta
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)

	modified:   libinih/inih (modified content)

no changes added to commit (use "git add" and/or "git commit -a")
$ git diff
diff --git a/libinih/inih b/libinih/inih
--- a/libinih/inih
+++ b/libinih/inih
@@ -1 +1 @@
-Subproject commit 5dbf5cb6b4027d5937726b8c499bd93c5b7d935d
+Subproject commit 5dbf5cb6b4027d5937726b8c499bd93c5b7d935d-dirty
```

c-mauderer, 2017-11-27:
OK. Odd. It seems that everything has been checked out correctly but somehow the content of `libinih/inih` got lost. You should be able to restore it by going into that directory and executing a
```
git co .
```
After that there should be the following files and directories in that folder:
```
cpp  examples  extra  ini.c  ini.h  LICENSE.txt  README.md  tests
```
I didn't see a bug in the build script just by looking at it. So if that problem doesn't occur again, I don't think that there is a chance to find out what happened.

doppioslash, 2017-11-27:
I tried the build script 4 times, and this happens every time. I'll upload the whole log in case it helps.

Doing ```git co . ``` from `grisp-software/libinih/inih` or even `grisp-software/libinih/` is not helping, the directory is still empty:
```
$ pwd
/[...]/grisp-software/libinih/inih
$ git checkout .
error: pathspec '.' did not match any file(s) known to git.
```

c-mauderer, 2017-11-27:
A complete log would be great.

doppioslash, 2017-11-27:
here is the log: https://gist.github.com/doppioslash/c7ec21140c657253960055b2bc4f6a74

c-mauderer, 2017-11-27:
The log seems to be OK except for the build failure in the libini part. But that part looks interesting:
```
Claudias-MacBook-Air:grisp-software doppioslash$ cd libinih/
Claudias-MacBook-Air:libinih doppioslash$ cd inih/
Claudias-MacBook-Air:inih doppioslash$ git checkout .
error: pathspec '.' did not match any file(s) known to git.
Claudias-MacBook-Air:inih doppioslash$ ls
Claudias-MacBook-Air:inih doppioslash$ pwd
```
The git submodule seems to be in a really odd state. I was only able to reproduce that output with the following steps:
```
> git clone https://github.com/grisp/grisp-software grisp-software-test
> cd grisp-software-test
> git submodule init
> git submodule update
# interrupt this during the handling of 'rtems' submodule
```
With that my inih submodule reached that odd state. In that case a `git status` in the `libinih/inih` subdirectory returned the following output:
```
…isp-software-test/libinih/inih 14:52 > LC_ALL=en_US git status
HEAD detached at 5dbf5cb
Changes to be committed:
  (use "git reset HEAD <file>..." to unstage)

        deleted:    LICENSE.txt
        deleted:    README.md
        deleted:    cpp/INIReader.cpp
        deleted:    cpp/INIReader.h
        deleted:    cpp/INIReaderTest.cpp
        deleted:    examples/config.def
        deleted:    examples/ini_buffer.c
        deleted:    examples/ini_dump.c
        deleted:    examples/ini_example.c
        deleted:    examples/ini_xmacros.c
        deleted:    examples/test.ini
        deleted:    extra/Makefile.static
        deleted:    ini.c
        deleted:    ini.h
        deleted:    tests/bad_comment.ini
        deleted:    tests/bad_multi.ini
        deleted:    tests/bad_section.ini
        deleted:    tests/baseline_disallow_inline_comments.txt
        deleted:    tests/baseline_multi.txt
        deleted:    tests/baseline_single.txt
        deleted:    tests/baseline_stop_on_first_error.txt
        deleted:    tests/bom.ini
        deleted:    tests/multi_line.ini
        deleted:    tests/normal.ini
        deleted:    tests/unittest.bat
        deleted:    tests/unittest.c
        deleted:    tests/user_error.ini

…isp-software-test/libinih/inih 14:52 >
```
So somehow interrupting the `git submodule update` left the repo in an odd state. Does that state is similar on your machine? In that case a `git reset HEAD .` and a `git co .` in the `libinih/inih` (**not** the one above that dirctory) should fix it.

doppioslash, 2017-11-27:
Indeed, it does fix it. This is odd.
I'm on the latest git from homebrew, 2.15.0, so it shouldn't be that.

c-mauderer, 2017-11-27:
Did you interrupt the first build process somewhere with Ctrl+C or something similar?

doppioslash, 2017-11-27:
No

c-mauderer, 2017-11-27:
Odd. So just to get it right: If you do a fresh clone (not only a cleaned up one) you get this behavior and the log that you attached a few posts ago?

doppioslash, 2017-11-27:
I shall wipe everything and try again, will let you know when it's done.

doppioslash, 2017-11-27:
It fails for unrelated reasons:
```
download: https://devel.rtems.org/raw-attachment/ticket/3057/0001-config-sparc-sparc.opt-mfix-ut700-New-option.patch -> patches/0001-config-sparc-sparc.opt-mfix-ut700-New-option.patch
download: https://devel.rtems.org/raw-attachment/ticket/3057/0001-config-sparc-sparc.opt-mfix-ut700-New-option.patch: error: HTTP Error 503: Service Unavailable
error: downloading https://devel.rtems.org/raw-attachment/ticket/3057/0001-config-sparc-sparc.opt-mfix-ut700-New-option.patch: all paths have failed, giving up
Build FAILED
  See error report: rsb-report-arm-rtems4.12-gcc-7.1.0-newlib-2.5.0.20170623-x86_64-apple-darwin14.5.0-1.txt
error: downloading https://devel.rtems.org/raw-attachment/ticket/3057/0001-config-sparc-sparc.opt-mfix-ut700-New-option.patch: all paths have failed, giving up
Build Set: Time 0:03:04.306061
Build FAILED
```

c-mauderer, 2017-11-27:
Hm, seems that devel.rtems.org is down. I get a "503 Service Unavailable - The server is temporarily unable to service your request due to maintenance downtime or capacity problems. Please try again later."

No Idea why the server is currently down but 503 is a temporary failure. So the server should be back soon.

doppioslash, 2017-11-27:
I'll try again later.

c-mauderer, 2017-11-28:
Can you close this bug if it doesn't happen again?