grisp/grisp-software #25: Built-in shell hangs after editing file

eproxus, 2017-11-28:
When editing a file on the SD-card the built-in RTEMS shell hangs aftewards. Steps to reproduce (using boot loader 1.0.1):

1. Create a SD-card with a `grisp.ini` file that points out a non-existing binary to boot from
2. Edit that file in the RTEMS shell started after boot fails:

   ```
   [/] # edit /media/mmcsd-0-0/grisp.ini
   ```

3. Save it and quit (<kbd>Ctrl</kbd>+<kbd>S</kbd> and then <kbd>Ctrl</kbd>+<kbd>Q</kbd>)
4. Try to see the changes:

   ```
   [/] # cat /media/mmcsd-0-0/grisp.ini
   ```

At this point, the shell freezes and does not recover. Before trying to edit or print the edited file, other commands seems to work (`ls` etc.).

kiwichris, 2018-06-22:
Does the shell hang if you exit the editor with out changing the file?

eproxus, 2018-06-22:
@kiwichris Didn't test that.

Ideally, this should be completely tested with a boot loader that is built with RTEMS 5 since it contains many fixes for the FAT file system.