grisp/grisp-software #26: Building grisp failing on Mac OS 10.10.5

doppioslash, 2017-11-28:
Following [instructions](https://github.com/grisp/grisp/wiki/Creating-Your-First-GRiSP-Application) `rebar3 grisp build` fails on:
grisp-software commit 649609a9e7deb3523577f841d0dad980537b4906
Mac OS X 10.10.5
Rtems 4.12
Erlang/OTP 19.3.6
Rebar 3.4.7
Grisp Toolchain 1.2-beta

Build log: https://gist.github.com/doppioslash/63e76918bf0eebef3670f4bef28dd20c

eproxus, 2017-11-28:
The problem is building with an old toolchain. The error `error: implicit declaration of function 'grisp_init_wpa_supplicant'` indicates that the Erlang version built uses WPA, but the toolchain version stated above doesn't have any WPA code. Try building with `master` of the toolchain instead.

doppioslash, 2017-11-28:
Trying with grisp toolchain master now...

doppioslash, 2017-11-28:
It works on master.

closed 2017-11-28 by doppioslash