grisp/grisp-software #28: Issue with the provided udev rules.

sylane, 2017-12-05:
The udev rules `build/99-grisp.rules` do not seem to work as expected on Linux.
I had to remove `ATTRS{bInterfaceNumber}=="01"` for the symlink to be created.

It seems that `ENV{ID_USB_INTERFACE_NUM}=="01"` would be doing what is wanted but I don't really understand why we would want to do that in the first place.

What are the reasons behind this part of the rule ?

c-mauderer, 2017-12-05:
The used FT2232 provides two interfaces. The first one (/dev/ttyUSB0 if no other device is connected) is used for debugging using OpenOCD. The second one (/dev/ttyUSB1) is the serial console interface. The ATTRS{bInterfaceNumber} was necessary to only create the link for this second port.

The rule has been written on an OpenSUSE 42.2 system with udev version 228-25.15.1. In my experience with other devices the ATTRS are not always consistent over different versions of udev. So it's quite possible that some adaption is necessary.

I'm not sure but it could be possible that ID_USB_INTERFACE_NUM has different values if there are more USB serial converters attached. Do you have any more information on how that value is generated?

sylane, 2017-12-05:
Ok, I see now.
I was using the rules without `ATTRS{bInterfaceNumber}=="01"` and it got linked to `/dev/ttyUSB0` and was not working. using `ENV{ID_USB_INTERFACE_NUM}=="01"` seems to work just fine now.

I am not 100% sure about where the value is generated, but it seems it comes from systemd, not udevd... so it may not be completly portable.

From some mailing list someone got around it not being defined by using the rule:
`SUBSYSTEMS=="usb", ATTRS{bInterfaceNumber}=="?*", ENV{ID_USB_INTERFACE_NUM}="$attr{bInterfaceNumber}"`

I am not sure what you are suggesting about issues with serial converters.

c-mauderer, 2017-12-05:
> I am not sure what you are suggesting about issues with serial converters.

I meant the following case:
- You have some random USB-serial converter connected that is registered as /dev/ttyUSB0
- Now you connect a GRiSP. That creates /dev/ttyUSB1 and /dev/ttyUSB2. Does the symlink point to ttyUSB1 or ttyUSB2.

But I just tested that. It correctly points to USB2. So I'll just create a patch with your suggested solution.

commit b2288217 (2017-12-05): build: Fix udev rules.

This fixes #28.
closed 2017-12-05 by c-mauderer