grisp/grisp-software #30: Debugging symbols missing when trying to debug the beam with gdb and supports script.

sylane, 2017-12-07:
I was trying to debug using openocd and using the gdb commands provided by OTP: `erts/etc/unix/etp-commands`.

The commands requires a set of symbols to be defined (the `etp_XXX` defined in `erts/emulator/beam/erl_init.c`) but they are not. I wonder if the issue is due to the compiler optimizing them out or any special cross-build issue.

peerst, 2017-12-07:
Hmm, depends what etp_XXX is for.  It might also be part of a build that uses threads or supports smp which we don´t do for default at the moment.

sylane, 2017-12-07:
They are defined in `erts/emulator/beam/erl_ini.c`, with the entry point of the beam... I don't see why we wouldn't have it.

sylane, 2017-12-07:
And looking at the generated object symbols, it contains a `erl_init` one...

peerst, 2017-12-08:
RTEMS uses a method where the compiler puts every function into its own section and then at the linking phase garbage collects sections that are not referenced.  This could cause functions to be removed if they are not referenced from the program.