grisp/grisp-software #31: Question: when building toolchain, is there a way to re-use build artifacts?

eproxus, 2017-12-11:
When running `./build/build-toolchain.sh` all tools seem to be always re-build. I.e. GCC takes hours to build every time the script is run. Is there a way to re-use the previous build artifacts and only re-build what has changed in some way? That would drastically reduce the build times, if possible.

c-mauderer, 2017-12-11:
I'm not aware of a possibility. But the rtems source builder has a lot of options. So either read the docs (https://docs.rtems.org/branches/master/rsb/index.html) or ask on the RTEMS mailing list.

By the way: Normally it shouldn't be necessary to re-build the toolchain. That has only been necessary during development when our RTEMS release changed frequently. It will again be necessary if we update to RTEMS 5. But otherwise there should be no necessity. Therefore the `build.sh` has a `--no-toolchain` option to skip that step.

eproxus, 2017-12-11:
Yeah, I get that. Just thought since some build tool runs in the bottom of this (Make? Waf? Both?) it would be a bit "smarter" 😄

closed 2017-12-11 by eproxus