grisp/grisp-software #44: Consider hosting rtems toolchain build-dependencies on GitHub

lwehmeier, 2018-06-20:
It appears that some patches the rtems toolchain build process depends on are hosted on an unreliable server. Maybe consider hosting them GitHub or removing them altogether as they appear to be irrelevant for the ARM toolchain build?
 
`grisp-software/build/build.sh`
```
[...]
config: tools/rtems-gdb-8.0.1-1.cfg
package: arm-rtems5-gdb-8.0.1-x86_64-linux-gnu-1
download: http://ftp.gnu.org/gnu/gdb/gdb-8.0.1.tar.xz -> sources/gdb-8.0.1.tar.xz
downloading: sources/gdb-8.0.1.tar.xz - 18.7MB of 18.7MB (100%)
download: https://gaisler.org/gdb/gdb-8.0.1-sis-leon2-leon3.diff -> patches/gdb-8.0.1-sis-leon2-leon3.diff
download: https://gaisler.org/gdb/gdb-8.0.1-sis-leon2-leon3.diff: error: <urlopen error [Errno 111] Connection refused>
error: downloading https://gaisler.org/gdb/gdb-8.0.1-sis-leon2-leon3.diff: all paths have failed, giving up
Build FAILED
  See error report: rsb-report-arm-rtems5-gdb-8.0.1-x86_64-linux-gnu-1.txt
error: downloading https://gaisler.org/gdb/gdb-8.0.1-sis-leon2-leon3.diff: all paths have failed, giving up
```

peerst, 2018-06-20:
Well thats not up to us at the moment:  We are using https://github.com/RTEMS/rtems-source-builder under the hood to build the toolchain and rtems.   Unfortunately it depends on many sites of which some are to available.  Having mirrors for these would be fortunate but I don’t know if these can be set up easily

kiwichris, 2018-06-20:
I moved the patch to the RTEMS servers yesterday. The RSB has been updated. Is the problem still present?

nextl00p, 2018-06-20:
@lwehmeier you could try building with branch `update-rtems-source-builder`, see #45 .

eproxus, 2018-06-21:
@kiwichris @nextl00p Just tried out the branch, it works fine and only rtems.org was contacted 👍

kiwichris, 2018-06-22:
Thanks for the report. I suppose this can be closed. :)

eproxus, 2018-06-22:
@kiwichris I've merged this to `master`, please try it out 🙂

closed 2018-06-22 by eproxus

lwehmeier, 2018-06-22:
Works fine on my Ubuntu 17.04 and Fedora 27 machines.
On Fedora I had to install the python2 development headers to successfully build the arm-rtems-gdb, maybe add that to the dependencies listed in the wiki?

nextl00p, 2018-06-22:
Python headers are already in the wiki: https://github.com/grisp/grisp/wiki/Linux-Development-Environment#fedora-27

Do you run latest rebar plugin?

lwehmeier, 2018-06-24:
Indeed, missed that one. I'm running the current GitHub master of the plugin:
```
{plugins, [
  {rebar3_grisp, {git, "https://github.com/grisp/rebar3_grisp.git", {branch, "master"}}}
]}.
```

eproxus, 2018-06-25:
@lwehmeier Are you saying that the `master` branch cannot build using the RTEMS 5 toolchain but the Hex release can? The Hex version is almost identical to the latest `master` in this respect...

lwehmeier, 2018-06-25:
I'm saying that once I changed my project's rebar config from
```
{plugins, [
  {rebar3_grisp, {git, "https://github.com/grisp/rebar3_grisp.git", {branch, "master"}}}
]}.
```
to 
```
{plugins, [
 rebar3_grisp
]}.
``` it became to build using the rtems5 toolchain. Maybe changing the plugin from git to hex triggered some updates/upgrades beyond what `rebar3 update && rebar3 plugins upgrade rebar3_grisp
` does? I'll try changing back to git master and report if the build breaks again or keeps working.

eproxus, 2018-06-25:
@lwehmeier Thanks for letting me know. I would expect the same behavior after upgrading the plug-in to the latest `master` version. `rebar3 plugins upgrade` should have upgraded to the latest `master` branch... Rebar's plug-in system is very adhoc so I'm not surprised there are potential problems here. A good sanity check is to completely remove `_build` and either verify or remove `rebar.lock` and then try again.

PR #45 (2018-06-22): Update rtems-source-builder to upstream master
Should fix #44 , but untested.