grisp/grisp-software #49: Missing Licence

breezeight, 2019-03-01:
I would like to use this project as a base on another opensource but there is no license on this repo.

breezeight, 2019-03-01:
I notice that other project in the Grisp org have a similar issue, some are under MIT, some are under Apache. May be using a single licence could benefit the project.

c-mauderer, 2019-03-01:
I think it shouldn't be too hard to get a license for grisp-software.
- The samples and applications as well as libgrisp already have a BSD 2 clause header
- Only 8 persons have worked on the build scripts. Most of it was done by me, one of the colleagues of @peerst colleagues or one of my colleagues. We should be able to get a OK for a license there.
- libinih could be a little tricky. That is an import from an external project. It has a BSD 3 clause license.
- The rest are subrepos with their own license.

So basically the question is: @peerst: You have the lead on that project so which license do you want? I would suggest BSD 2 clause for all code in grisp-software (except for libinih and the subrepos) like already used in the samples.

I think a single license for the other repos could be a little difficult. I'm only active in the RTEMS parts but that is an external project. So the license can't be easily changed (although RTEMS works towards a BSD stile license). Some of the other repos most likely have the same problem.

peerst, 2019-03-20:
BSD licence would be fine for me, slight preference for Apache 2 since our Erlang code and Erlang/OTP itself.

But if BSD is easier thats fine by me

c-mauderer, 2019-03-22:
I would slightly prefere BSD license because most of the code already has it. As far as I know BSD-2-clause should be compatible with Apache 2. So you shouldn't get any problems there. See http://www.apache.org/legal/resolved.html#category-a

Below is a list of all authors (using `git shortlog -e`). Also most commits are done by me, I think we should collect a OK of all of them in this ticket.

- 7 Adam Lindberg (@eproxus)
- 5 Andreas Leibold (as far as I know, he has no github account - I'll collect that one as an Email)
- 161 Christian Mauderer (@c-mauderer)
- 3 Kilian Holzinger (@nextl00p)
- 7 Peer Stritzinger (@peerst)
- 11 Sebastian Huber (@sebhub)
- 3 Sebastien Merle (@sylane)
- 6 nextloop (@nextl00p again?)

As soon as we have green light, I'll add the following text as LICENSE:

````
There are a few imported paths in this repository. Namely:

- libinih/inih: New BSD license (see libinih/inih/LICENSE.txt)

- rtems, rtems-libbsd and rtems-source-builder: See the Licenses of the
  subrepositories.

For all other files the following bsd-2-clause is applicable:

   Redistribution and use in source and binary forms, with or without
   modification, are permitted provided that the following conditions
   are met:
   1. Redistributions of source code must retain the above copyright
      notice, this list of conditions and the following disclaimer.
   2. Redistributions in binary form must reproduce the above copyright
      notice, this list of conditions and the following disclaimer in the
      documentation and/or other materials provided with the distribution.
   
   THIS SOFTWARE IS PROVIDED BY THE AUTHOR AND CONTRIBUTORS ``AS IS'' AND
   ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
   IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
   ARE DISCLAIMED.  IN NO EVENT SHALL THE AUTHOR OR CONTRIBUTORS BE LIABLE
   FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL
   DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS
   OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION)
   HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT
   LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY
   OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF
   SUCH DAMAGE.
````

peerst, 2019-03-24:
Green Light ;-)

Thanks,
— Peer

eproxus, 2020-03-09:
@nextl00p @sylane Do we have your permission to use the BSD license for your commits?

nextl00p, 2020-03-09:
I agree to license my work on this project under the BSD 2-clause license.